CymatiCorp/CyKit #1: Linux Error

ringalls20, 2017-02-18:
Hello,

I am trying to run this code on a Raspberry Pi 3 model. 
I keep getting this error:
>Linux detected.
>Traceback (most recent call last):
>  File "/home/pi/folder/emotiv.py", line 674, in <module>
>    a.setup()
>  File "/home/pi/folder/emotiv.py", line 404, in setup
>    self.setup_posix()
>  File "/home/pi/folder/emotiv.py", line 508, in setup_posix
>    serial,hidraw_filename = get_linux_setup()
>TypeError: 'NoneType' object is not iterable
 
Any suggestions would be appreciated!

warrenarea, 2017-02-20:
yes, i intended CyKit to be primarily ran from the Windows
environment, as I didn't always have a Linux setup to test with, 
and could not ensure the functionality.  So I believe I removed
much of the Linux support from emotiv.py 

which was originally part of the emokit build.
so its possible you might be able to take the emokit 
setup and apply it here...

although to complicate matters, I think emokit might have updated
their version to support Python 3, so you would have to find an older
Python 2.7.6 emokit build as well... 

may I ask what you what version of linux you are running on the raspberry pi, 
and what you intend to do with it on that setup?

ringalls20, 2017-02-21:
I am running the latest OS Debian Jessie with current updates.
I am intending to use this code to extract data from the Emotiv Epoc using the Pi. This is part of a Automated Meditation Lamp in which the Pi is the processor of the design and makes output to the LEDs.
This makes for a nice road block in Te project but of course an opportunity to learn something new.

warrenarea, 2017-02-21:
Well, i tried to install that latest jessie build onto my arduino pi controller,
however I can't seem to get the device to turn on (possibly a power cord issue).

So it might be some time before I can possibly adapt Cykit to work with that setup.

CyKit was designed primarily as a network server interface, so that it can stream the
raw EEG data to another program/client.

So I'm not entirely sure if this setup would be what you are interested in, 
the openyou / emokit release on github, is already set up for Linux and is maintained
pretty frequently though, so this might be a better option for you.

https://github.com/openyou/emokit

I might add Linux support to mine if I can get my Pi to power on, but that might be
some time.

warrenarea, 2017-02-21:
This kind of reminds me of what you're after 
https://www.youtube.com/watch?v=HW5G7_zfi4E&t=3m21s

closed 2017-08-30 by warrenarea

konanast, 2019-02-20:
Hello,
This is a great project, I would prefer to use this one instead the old emokit.
Is there anything new about linux support? or a workaround?
I am trying to run it on Ubuntu 18.04.

weronikamichalska, 2019-08-28:
> Hello,
> This is a great project, I would prefer to use this one instead the old emokit.
> Is there anything new about linux support? or a workaround?
> I am trying to run it on Ubuntu 18.04.

Any info?