CymatiCorp/CyKit #3: TypeError: 'bool' object has no attribute '__getitem__'

Athif-UoM, 2017-08-30:
Hi

I have set up the CyKit and followed the read me. When I run the examply.py the following error message is thrown. I understand that this should be a problem with my serial number because the serial number is apparently taken in as bool (If you see the final error message) and therefore it throws a type error. Could you please enlighten me what could be the problem

Thanks 

Windows detected.
Product name Receiver Dongle L01
device path \\?\hid#vid_1234&pid_ed02&mi_00#8&97ab72&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
instance id HID\VID_1234&PID_ED02&MI_00\8&97AB72&0&0000

Product name Emotiv RAW DATA
device path \\?\hid#vid_1234&pid_ed02&mi_01#8&2902593b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
instance id HID\VID_1234&PID_ED02&MI_01\8&2902593B&0&0000
Devices Selected: 0
Press Enter to continue...

  File "A:\Internship\emotiv EPOC\CyKit-master\Python\emotiv.py", **line 262, in is_old_model
    if "GM" in serial_number[-2:]:
TypeError: 'bool' object has no attribute '__getitem__'**
Wed Aug 30 16:04:15 2017 <Greenlet at 0x2a49c10: <bound method Emotiv.setup_crypto of <emotiv.Emotiv object at 0x02A65A70>>(False)> failed with TypeError

warrenarea, 2017-08-30:
your output indicates that the Devices Selected is 0

which means that the USB device was not selected for one reason or another.

I think its Line 422 and 423:
 if device.vendor_id != 0x21A1 and device.vendor_id != 0xED02:
   continue

Try deleting those 2 lines.

Considering its not prompting you to confirm the device, that is likely the problem,
I think it was designed to check for specific vendor id of the usb device. 

Since we're using the product_name anyways,  In my current code, i found these unnecessary,
 and deleted them.

and since there is no device selected, it is not able to get a serial number, why the
serial_number reference is actually triggering the error.

Athif-UoM, 2017-08-31:
Thanks for pointing out. It worked (I think) because the output looks like this 

Emotiv RAW DATA Found!

Use this device? [Y]es? y

 Devices Selected: 1
Press Enter to continue...
True
0x39 
0x00 
0x35 
0x54 
0x34 
0x10 
0x36 
0x42 
0x39 
0x00 
0x35 
0x48 
0x34 
0x00 
0x36 
0x50 

I tried looking in the documentation how to see the table of sensor values plotted in the command line (like in Emokit). Could you also point me where to look for a detailed instruction to obtain signal values?

Thanks a lot in advance !

warrenarea, 2017-08-31:
yes, it appears you got it working. Well done.
This is actually helpful to me, because it proves that the "Emotiv RAW DATA"
is for the Epoc model.

the 0x39 0x00 etc... that is just hexadecimal for your USB's serial #
if converted to ascii, it would probably look "something" like 'SN05120081241GM'
SN for serial number, then a purchase date, and then a unique ID plus 'GM' for Epoc model.

I suggest looking at my other Python examples, especially the one where it retrieves
the data from the packet. like Cyos.py i think.

for learning, what I suggest, is try using the 'print' feature as much as possible, 
and try printing out as many variables as you can, like... print "serial" + self.serial_number
in languages like Python, debugging with 'print' commands are pretty much a must
in figuring out what is going on behind the scenes, and how things are running.

Python is an amazing language, in that... it teaches people how to write 'clean code', 
where code is neat and organized. 

Python also can be a very frustrating language, in my experience, because quite often
you will get an error in one place, and the problem resides in a completely different
area.  Until you learn how it handles things, trial and error is hugely important. 

Another recommendation, is to learn about the  try: and except commands.
as it lets you "try" a piece of code, and if it fails, it sends a more detailed message about
why it failed.   Which is very helpful in isolating problems with your code.... as well as
creating workarounds.

Python 3+ i have found even more frustrating than Python 2.7, because it requires you
to be more specific with your strings.  Which is a nice concept, but can drastically slow
you down learning about them.  There is also no improvement in speed either.  so there
is no motivation for me to change, other than possibly it being maybe more secure.

I think my new version of CyKIT will be even more helpful to you... but I don't want to
release it just yet, because I want to make it as best as i possibly can.  I think it will help
to open up a few new ways for people to work with their Emotiv devices.

Athif-UoM, 2017-09-11:
Hi

Thanks for the help. I think it works. I was able to see output displayed after setting up the OpenVibe acquisition client. Looking forward to try the new version of Cykit.

Thanks for the help again!

closed 2019-01-23 by warrenarea