CymatiCorp/CyKit #4: The units of voltage values imported into OPENViBE

Athif-UoM, 2017-08-31:
When the signal acquired from  Cyckit is used as an input to the OPENViBE, the plotted signal range seems to be the usual EEG (-100 to +100). However the documentation says that the values imported are signed integers with range (-8192 to 8192). Is there built in conversion from bits to uV in the acquisition? Do I have to manually correct the voltage range.

Thanks for the help

warrenarea, 2017-08-31:
The conversion should already be taken place with the get_level function.

OPENViBE i think just auto-scales the values from that point.

The values from the headset will be in the range of -8192 to 8192

One thing I usually do when programming the display output for the EEG Signals,
is to take samples every now and then.... average the results, to create a 'baseline' 
then subtract that baseline from the signal.  OPENViBE does this already for you,
but it should probably be implemented in CyKIT as well.

Keep checking back at this CyKIT repository, because I am currently working on 
a second version of CyKIT that has some really great features.

Athif-UoM, 2017-09-11:
Great! This really helped. I will keep in touch here. Thanks a lot !

closed 2019-01-23 by warrenarea