CymatiCorp/CyKit #5: The executable listens on port 55555

725640, 2017-09-06:
When running the Cyos executable it is listening on port 55555 ? 

And I believe it should just be 5555 ?

warrenarea, 2017-09-06:
yeah, that was a typo, it defaulted to 5555, when it said it was 55555.

I never really paid the default connection too much attention, because, you can specify
your own port to use, which is what I did... and the batch files specified the correct port, 
perhaps why no one had any problems with that before.

I edited it in the code.  good eye.

Keep checking back at this repository.... I plan to release an entirely new setup for CyKIT,
it will have support for each Emotiv model, epoc, epoc+ and insight. 

I rebuilt the connection setup, and the threading support is much better, and cuts out the 
dependency on gevent and greenlet.  plus it has a whole new interface.

its about half finished, thinking about releasing it early, just so people can start making 
use of it.

725640, 2017-09-07:
The batch files also did the same thing, so in Cyos.py I changed the port to 5555 instead of sys.argv[2] and then it manged to connect. 

It is quite strange how it hasn't caused any issues for anyone else, maybe something weird is going on with my system. But thank you, this is great !

warrenarea, 2017-11-07:
https://github.com/CymatiCorp/CyKITv2

closed 2019-01-23 by warrenarea