CymatiCorp/CyKit #6: It seems that product_name needs updating

lidh15, 2017-09-19:
Several days ago I bought a new EPOC+. When I use CyKit to get the data, there is no device connected. It turned out to be the fact that hid returned an unseen string ‘EEG Signals’ as device.product_name. After adding this to the possible product_name, I was able to get some data. I still need to check if the data is valid.

warrenarea, 2017-09-19:
i will save you some time, the data is not currently valid, because the encryption key is not
correct for the Epoc+

I am releasing, soon, a new version of CyKIT that will take care of all these issues. 
So keep checking back here, for the newer version.

lidh15, 2017-09-24:
I'm also interested in the modified hardware you've shown because the distribution of the electrodes doesn't match my application. The devices I've used before all had shielded wire for the electrodes in order to get rid of Interference signal, but EPOC didn't. I wonder how this works.

warrenarea, 2017-09-24:
I think the whole principle behind EPOC, is that it relies on the reference sensors heavily
to take care of interference, or negate the interference...

Because all of my sensors were so compacted together, and they were in such close 
approximation to the battery/control circuit/eeg circuit,  i thought it would be prudent
to shield them by using aluminum foil.  So I put layers of it in between the circuits, and
also on each individual wire.

I think this is probably why Emotiv desigend their circuits to sit in the center of the headset,
is that it is just far enough, that it might not interfere with the sensors.

Either way, I still think the foil is a good idea. The headsets provided by Emotiv are not really
designed to be very robust, or offer many safeguards.  They have a decent design and throw
it all together and offer it at a reasonable price. 

Judging by how easily the parts fall apart internally, I would not say that as much care is put
into the assembly of these, that could be.  

If your headset is still intact though, I would recommend not changing it.   and also... unless you
have a specific application in mind, the modification i made, is most often not a desirable one, because
each sensor is receiving almost the same neural data. 

if i didn't have the epoc+ and insight to experiment with at the moment, I would probably redesign
my Epoc, but it just makes sense right now to keep it in 1 functional piece, so that I can continue 
testing with it.

lidh15, 2017-09-25:
Well, I have done a little harm to the device, but I will take your advice and just stop here.

ivankras, 2017-09-25:
Hi! I'm really enthusiastic about getting the EPOC+ to work with CyKit. I'm using it for a college project, so it would be great. I'll wait for the new version; meanwhile, I'm understanding the code. Anyway I can help, just tell me, @warrenarea

warrenarea, 2017-09-25:
well, i was looking to add all of the features to make it a complete demonstration.

However, I think I've got enough complete, I should probably just release it and then continue 
updating it.  That way, someone might be able to utilize it, while I'm still working on it, and at
very least, see what I have going on, and make suggestions.

ivankras, 2017-09-25:
Awesome. Then I'll wait for you to release it. Cheers!

warrenarea, 2017-11-07:
https://github.com/CymatiCorp/CyKITv2

closed 2019-01-23 by warrenarea