CymatiCorp/CyKit #7: Importing Channel Names to OpenVibe

Athif-UoM, 2017-10-04:
Hi,

When the acquisition server of OpenVibe is connected to Cykit following the instructions on Readme, 14 channels labeled 1-14 are correctly extracted and shown. However, I want to know what these channels are. For example Emotive has the following channel locations. 

[image]

How do these channel locations map to channel 1, channel2, ... channel14 in the signal stream imported to OpenVibe. I am very thankful if you could point me the direction

Thanks a lot in advance.

warrenarea, 2017-10-04:
If you look at the code you will see the string of contact sensor names:
AF3 F7 F3 FC5 T7 P7 O1 O2 P8 T8 FC6 F4 F8 AF4

In openvibe, it will appear in that order, ie:
Channel 1 = AF3
Channel 2 = F7
Channel 3 = F3
and so on.

This is for the Epoc model.

Athif-UoM, 2017-10-05:
Thank you very much. It did not occur to me to check the codes and am sorry for the trouble you had to go through. Thanks again !

warrenarea, 2017-10-05:
its no trouble. 
I've been crawling over this code for days/months, so its all on the tip of my brain. lol
I am making tons of progress on the 2nd version of CyKIT... I think everyone may be impressed.

I would like to release it, but I want everything to be perfect. 
I'm also kind of wondering if Emotiv will frown on what I've been working on.

lidh15, 2017-10-09:
The most important function in my opinion is EEG logger which is not open in Emotiv SDK. And Emokit just stopped in 2016. As long as the logger is available, CyKit is perfect in my eyes.

Athif-UoM, 2017-10-18:
Hi,
Its amazing work you are doing! I can't wait till you release the version that supports EPOC +. Around how long will it be before its out? 
Thanks !

warrenarea, 2017-10-23:
I apologize for the delay in response.  I plan to have something released by the end of this week.

The reversing process has proven a bit more difficult than anticipated.  Their software designer is
fairly proficient in obfuscating his code, and I'll have to admit my reverse engineering skills have
much to be desired.

lidh15, 2017-10-24:
You know what, if you had said "release in two weeks" a week ago, I wouldn't spent $1k on EmotivPro.
Now I tell myself "Calm down, $1k is not too much, just imagine you lost your iPhone X in the crowd."
That is life. Wubba Lubba Dab Dab

warrenarea, 2017-10-24:
Well, as it stands, I have not deciphered any epoc+ raw data functions, and it will probably still be
some time before they are sorted properly.  I think you'll just have to wait and see what I have set up.

Also, I'm not sure if you realize it or not, but if you purchased the license to run EmotivPRO, you 
also purchased a license seat that allows you to run an unlimited number of raw data accesses to 
the emotiv community SDK.

Just open up your C:/users/administrator_name/appdata/EDK/Ecvn.bin file, and place that license
key into your CommunitySDK/examples_extra_prime/Python/ActivateLicense.py file.

Before each time you want to use the logger, you will have to run the ActivateLicense file, and
enter 1 debit.

warrenarea, 2017-10-24:
around line 65-66 of that activation file, you will want to put your Emotiv username/password
and then around line 140, you put your Ecvn key.

lidh15, 2017-10-24:
Got it. And I find that Emotiv only provide the software for Mac x64 and Windows x86, but all of my computers are Windows x64 thus I have to use a lot of patches. I hope your work are stable for all architectures.

warrenarea, 2017-11-07:
https://github.com/CymatiCorp/CyKITv2

closed 2019-01-23 by warrenarea