CymatiCorp/CyKit #9: The selected device is always 0

Athif-UoM, 2017-11-25:
Hi, I have previously installed cykit and used it with OpenVibe with my windows 7 computer with python 2.7.11. However when I followed the same procedure on another computer with windows 7 and python 2.7.14. I cannot get the device selected to be 1. So when I proceed with thats I can only get the values printed as 0s for all the signal values. I am attaching a screen capture of the error (top right) of which I cannot understand the cause. I will be much thankful if you could point the correct direction. 

[image]

warrenarea, 2017-11-25:
try taking out line 422 and line 423 of CyKIT/emotiv.py

i'll remove it from the code on the repository as well.

See if that works, i think it blocks some devices from being detected.

warrenarea, 2017-11-25:
This is the lines i removed:
```
if device.vendor_id != 0x21A1 and device.vendor_id != 0xED02:
                    continue
```

Athif-UoM, 2017-11-25:
Thanks, but following your suggestion to an earlier issue, I tried removing the lines already and it is still not working.

I went through the code and am suspecting if it has something with hidapi. Even if the device named 'Emotiv Raw Data' is found (The last of the three devices printed in cmd). For some  reason it is not selected. That's puzzling for me. 

Would be much appreciative for your opinion on this

warrenarea, 2017-11-25:
well, for some reason i don't think so....

because hidapi "is" detecting the device exists....

are you sure that the "Emotiv Raw Data" was the device that needs to be used?

you "might" try adding a new entry for 'Receiver Dongle L01' 
its worth a try.... but if it worked for Emotiv Raw Data, then.... i'm not sure.

warrenarea, 2017-11-25:
try removing the confirmation for the "Emotiv Raw DATA"

or try adding letters around the Emotiv RAW DATA, to see if there is a space or something..

i think if you play around with it, removing stuff and having it print out as much
information as you can, you might find something....

Athif-UoM, 2017-11-25:
Thanks ! I will get back with what I get after playing around there

warrenarea, 2017-11-25:
sounds good.

warrenarea, 2017-11-25:
not really relevant to your connection problem....
but your connection port is different. lol

Athif-UoM, 2017-11-25:
Pardon me for the ignorance. Could you please explain me on that. It could be the reason as well  (?) :O . Is this the difference? 
[image]

warrenarea, 2017-11-25:
yeah, thats not relevant. 

your issue is your device isn't being added, its obviously found. 

and what you have circled is a localized port, basically a secondary port that windows
is using kind of behind the scenes to establish your connection.  pay no attention to it.

Athif-UoM, 2017-11-25:
Thanks. I do not have the access to the computer with the problem at the moment. I was trying to recreate the problem using my computer instead and interestingly, uncommenting line 422 and 423 does that exactly. I will update as soon as I get access to the problematic computer.

closed 2019-01-23 by warrenarea