CymatiCorp/CyKit #10: A question about sample rate

imyzx2017, 2019-01-23:
When I chose EPOC+(Research) model or EPOC+(Standard) model, the data sample rate is supposed to be 256Hz(Mentioned in ./eeg.py Code).

But the result I got is 128Hz data with 64Hz gyro data.
So what should I do to access 256Hz data? Does Cykit support high sample rate?

Thanks!

warrenarea, 2019-01-23:
choosing which key model you have, (ie: EPOC+ (Research) model or EPOC+ (Standard) model)
determines which encryption type to use, for your particular headset model.

Research/Standard is the older names we used for py2.7.6,  you should upgrade to python 3
which uses the names Premium/Consumer model.   

Either way, most users are going to have Consumer/Standard model, so you should probably disregard
using the Premium/Research keymodel, as it won't be relevant here.

(For the latest Python 3 version)
keymodel 6 is used for when the EPOC+ is in its (16-bit mode) 128hz or 256hz mode.
and keymodel 7 is used for when EPOC+ is in its (14-bit mode) (ie. the old EPOC data mode)

for the latest Python 3 version, 
you plug your headset directly into the USB cable,
turn the device on (the white light MUST be on)
and run CyKIT as you normally would streaming.
i also recommend using the config option "pywinusb"

you are then given a menu, that allows you to change what "mode" the device will 
stream the data in, changing from 128hz mode or 256hz mode, as well as changing
what gyro speed is.

unplug your device from usb, and it should now stream in the mode you
selected.  you can verify the device is streaming properly to the mode set,
by using the config option "outputdata" and verifying the COUNTER is going
from 0-127 or 0-255

you can check the discord chat link, for further assistance.

warrenarea, 2019-01-23:
Also see: [How to Change EPOC+ hertz modes](https://github.com/CymatiCorp/CyKit/wiki/How-to-Change-EPOC(plus)--modes)

closed 2019-01-23 by warrenarea