CymatiCorp/CyKit #12: Device compatibility of running the CyKIT.PY

AnPang940, 2019-03-10:
my device: WIN 10 EDU
when I run the CyKIT.py in cykit-master（py3）and the error comes out as it can not be runned in the computer and access denied. 
How can I deal with it? Can you give me some advice?

yoyou446, 2019-08-07:
No Linux support?

samaa985, 2020-01-18:
> my device: WIN 10 EDU
> when I run the CyKIT.py in cykit-master（py3）and the error comes out as it can not be runned in the computer and access denied.
> How can I deal with it? Can you give me some advice?

i have the same problem do you find any solution???

warrenarea, 2020-01-20:
no linux support at this time.

@samaa985 you might try running CyKIT.py without any config options. (parameters)

i've had an issue with CyKIT in the past, or rather, an issue with Windows 10,  where...
if you try to pass a string of arguments to it,  Windows "smartscreen" will kick in, and 
will even go so far as to delete Python.exe     because i guess it gets flagged as trojan-like
behavior, because of all of the command arguments. 

that doesn't sound like the issue you're having here though,  sounds like your user account
for the operating system might not have the appropriate access/permissions to run the program.  
unfortunately "access denied" gives you little information as to what is being denied.... 

only thing i can think of at the moment, is that it opens a USB device, and maybe it doesn't have
the permissions to allow that.  a short-term solution would be to run it as administrator.  
i would be interested in discovering what is causing this issue, if you could provide more details.

Got to love Windows error messages.... "Access Denied" is just as ambiguous as their 
"Something happened" error.    As a developer, i don't see how you could feel good about
that sort of erroneous ambiguity.

warrenarea, 2020-01-20:
why if you look at my code, you will notice my error messages (for the try code blocks)
all start out with the function name, and a unique number.  That way, it makes tracing
back the error a simple matter.

By the way, there are some DLL's that get loaded,  they're from the pycryptodomex repository,
anyone can compile them, but the compilation process is very fragile (in my opinion),  can take
a good chunk of your time to figure out what compiler versions to use and what settings to make
it work.   Why i precompiled them into the project so that other people could get on with their
work and not have to bother with that nonsense.  It's saved me a ton of troubleshooting time as
well,  because i don't have to worry about people setting it up wrong, or using wrong versions, etc.

Did the same with the bluetooth drivers,  i didn't compile those, but i trust the source of them, and
the compilation code is included as well. 

i suppose it's possible the "access denied" could come from the attempts to load the DLL's,
why i bring them up.