CymatiCorp/CyKit #14: Archiving emokit and pointing README to CyKit?

qdot, 2019-07-22:
Looks like Cykit has come a long way since I last looked! Emokit is going to continue to stay dead (I have zero time to even think about EEGs anymore), and I was considering archiving it, so I should I just point people over here in the README?

warrenarea, 2019-07-22:
it's funny you bring it up, i was thinking of emokit/cykit earlier today as
well.

i realize i said i was going to work on emokit, and it just never
materialized.

my plan has always been... that CyKIT would be a generic streaming via TCP
,  for openvibe and browser.
and then emokit, could be the modularized version that people could drop
into their projects.

Hopefully I will have a bit more time (and energy) to finally put out my
next version of CyKIT,  which will be
much more organized than it is now, and has some pretty cool new features.

I lost my steam on the project though... and it sometimes weighs on me,
that one day Emotiv may not be so
pleasant about these projects. (because of how far i've run with them)

i kind of realized though why bill schumacher deleted his github account,
maybe he was just trying to distance
himself from this project.  It's kind of difficult though when... you feel
like you're on the right track... but if you
are working toward something you can't profit from,  it can be sometimes
difficult.

i think i will have to wait until my next release, before i finally decide
where to go with this.
what do you think?

qdot, 2019-07-22:
Yup, no rush on my end, was just doing some cleanup and had this thought, figured I'd get it out there while it was still fresh.