CymatiCorp/CyKit #15: Version for Mac and Linux

avolution, 2019-10-09:
Would be awesome when you guys could port/modify in a way that it will also work on Mac and/or Linux.

josuuribe, 2019-11-08:
I am working on it using this repo as start point https://github.com/hugofloresgarcia/CyKit, but I will need support from @warrenarea  or any other that has knowledge about the code, if anyone is interested please contact me.

warrenarea, 2019-11-08:
many resources can be found on the CyKIT discord pins section. 
i believe there was a couple people already who have created OS X versions,
as well as a fork working toward the linux distribution.  you should try
searching the discord for more info.

josuuribe, 2019-11-11:
I have updated CyKit for Linux but i can no push to repo, you can go to Discord or contact me

warrenarea, 2019-11-11:
Until i can take the time to review all of the material and integrate it into CyKIT,
i have put a referring link to your Github branch, regarding Linux/Mac support. 

that is the best i can do at this time.

chewxy, 2020-03-27:
https://github.com/hugofloresgarcia/CyKit is 404 ing for me.

warrenarea, 2020-03-27:
yes.  they dropped the repo.  makes things a little difficult for us.
removed it from readme.

iPsych, 2022-12-27:
@josuuribe Do you have the dropped repo of hugo?
Any advance in MacOs porting?