CymatiCorp/CyKit #17: How do I open a python file in cykit？

wh1233, 2020-04-18:
Hello,I want to see the code in cyPyWinUSB/winapi.py.Why can't I open it?

warrenarea, 2020-04-18:
not sure, seemed to open fine for me.

wh1233, 2020-04-19:
Thank you for your answer.
We're working on a BCI project using EPOC+ and we want to save some raw EEG data ,but I don't know yet how to connect this device like in your project,how do I get the port number and what is the data format of the device？
Thanks again

warrenarea, 2020-04-19:
https://cdn.discordapp.com/attachments/415564084650442762/696134513969070110/tests.PNG

you can use a command line like the one listed in this picture

and then you can connect using chrome/firefox browser,  opening \Web\CyKIT.html 
and clicking connect button, once the server is running.

wh1233, 2020-04-20:
Thanks, cykit is very practical and convenient. Due to the needs of our project, I will learn some procedures of connecting emotiv epoc+. What type of data is initially obtained when connecting to the device? How can the data be divided into 14 sensor channels? Is there any protocol? Would you please give me some guidance?
Thank you very much！