CymatiCorp/CyKit #19: Streaming Emotiv data via CyKit3

teknomage, 2020-05-30:
Hello Warren, it's been a couple of years since I last reached out to you about a lagging issue that I was experiencing with the previous CyKitv2 (see the original thread here: [(https://github.com/CymatiCorp/CyKITv2/issues/9)] )
Well, I'm glad to say that after I upgraded to this newer CyKit version with Python 3.4 (I couldn't get it to work with Python's higher versions), that lag in the data display has reduced significantly.

Now, the question I have is: it's not really an issue but I'd like to know how I could stream the live incoming EEG data from Emotiv via CyKit into a python program. I understand it's possible to stream to OpenVibe (and even Matlab?), but all I need is a simple python listener script to receive the data. I'm not an expert in python, so if you could provide me with an example snippet or point me in the right direction, that would be most appreciated!

Thanks for all the great work you've done here and for keeping this CyKit project alive, love the updated interface too! :)

warrenarea, 2020-06-07:
Added "Examples" to the repository you can take a look at
for how to retrieve the data directly from the CyKIT engine.

I have plans to improve upon this system, so this is just the 
current available method.

teknomage, 2020-06-07:
That's great, Warren! Thank you so much for putting this up!

closed 2022-09-19 by warrenarea