CymatiCorp/CyKit #20: Emotiv Insight

blairzaye, 2020-06-06:
Hey There

I have followed you How-to-Install-CyKIT'

https://github.com/CymatiCorp/CyKit/wiki/How-to-Install-CyKIT

and get to Step 5. and get... 

C:\CyKit-master\Py3>c:/Users/Blair/AppData/Local/Programs/Python/Python38-32/python.exe .\CyKIT.py
Traceback (most recent call last):
  File ".\CyKIT.py", line 16, in <module>
    import eeg
  File "C:\CyKit-master\Py3\eeg.py", line 161, in <module>
    from cyCrypto.Cipher import AES
  File ".\cyCrypto\Cipher\__init__.py", line 3, in <module>
    from cyCrypto.Cipher._mode_ecb import _create_ecb_cipher
  File ".\cyCrypto\Cipher\_mode_ecb.py", line 40, in <module>
    raw_ecb_lib = load_pycryptodome_raw_lib("cyCrypto.Cipher._raw_ecb" + cyVersion, """
  File ".\cyCrypto\Util\_raw_api.py", line 259, in load_pycryptodome_raw_lib
    raise OSError("Cannot load native module '%s': %s" % (name, ", ".join(attempts)))
OSError: Cannot load native module 'cyCrypto.Cipher._raw_ecb38x32': Trying '_raw_ecb38x32.cp38-win32.pyd': Could not find module 'C:\CyKit-master\Py3\cyCrypto\Cipher\_raw_ecb38x32.cp38-win32.pyd' (or one of its dependencies). Try using the full path with constructor syntax., Trying '_raw_ecb38x32.pyd': Could not find module 'C:\CyKit-master\Py3\cyCrypto\Cipher\_raw_ecb38x32.pyd' (or one of its dependencies). Try using the full path with constructor syntax.

any ideas? I really dont know python at all... 

I am on a macbook 2020 in win 10 with boot camp with a Emotiv Insight with dongle, trying to get OSC into touchdesigner. most interested in the emotion performance metrics, but would also like the raw data and the brainwaves (alpha, beta, etc)

I downloaded the latest python 3.8 but maybe I need to try with 3.7? anything else i need to try?
I havnt been in MS-DOS or Command prompt for years.. Been hiding in MacOS since 2012 :)

blairzaye, 2020-06-06:
ah yes... just had a dig into the folders as per above and obvs it doesnt have 'C:\CyKit-master\Py3\cyCrypto\Cipher_raw_ecb38x32.cp38-win32.pyd'

blairzaye, 2020-06-06:
ok! so it works with 3.7 :) 

but..

now I am stuck here 

C:\CyKit-master\Py3>
C:\CyKit-master\Py3>c:/Users/Blair/AppData/Local/Programs/Python/Python37/python.exe .\CyKIT.py 127.0.0.1 7000 4 info+confirm+format-3
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)
> USB Device (No Additional Information)

> Found EEG Device [EEG Signals]

 Use this device? [Y]es? y
══════════════════════════════════════════════════

> Listening on 127.0.0.1 : 7000
> Trying Key Model #: 4
(-) Connecting . . .

I believe mine is 4 - Insight (Consumer Model) but mine just hangs... any ideas? 

I dont understand why there is the 'Listening on 127.0.0.1 : 7000' if its connected on the dongle. Is this for communicating to the Insight? 

also i see there is 'generic'             Connects to any generic program via TCP. (Can be used with other flags.) so im guessing I can use this some how instead of OSC?

blairzaye, 2020-06-07:
ah ha! went over to the discord got it working on the CyKIT.html.. trying to work out the performance matrix now..

warrenarea, 2020-06-07:
I just pushed the  necessary `pycryptodomex` binary files necessary to get Python3.8 working.

plusmoreira, 2020-06-13:
Hi guys!

I've just downloaded the master branch today and got Python 3.8.3 installed on Win10 - 64bits.

Tryied to run  CyKIT and got the same error as @blairzaye.

```
C:\CyKit-master\Py3>python .\CyKIT.py
Traceback (most recent call last):
  File ".\CyKIT.py", line 16, in <module>
    import eeg
  File "C:\CyKit-master\Py3\eeg.py", line 161, in <module>
    from cyCrypto.Cipher import AES
  File ".\cyCrypto\Cipher\AES.py", line 70, in <module>
    _raw_aes_lib = load_pycryptodome_raw_lib("cyCrypto.Cipher._raw_aes" + cyVersion,
  File ".\cyCrypto\Util\_raw_api.py", line 259, in load_pycryptodome_raw_lib
    raise OSError("Cannot load native module '%s': %s" % (name, ", ".join(attempts)))
OSError: Cannot load native module 'cyCrypto.Cipher._raw_aes38x32': Trying '_raw_aes38x32.cp38-win32.pyd': Could not find module 'C:\CyKit-master\Py3\cyCrypto\Cipher\_raw_aes38x32.cp38-win32.pyd' (or one of its dependencies). Try using the full path with constructor syntax., Trying '_raw_aes38x32.pyd': Could not find module 'C:\CyKit-master\Py3\cyCrypto\Cipher\_raw_aes38x32.pyd' (or one of its dependencies). Try using the full path with constructor syntax.

```

I didn't get what was the solution just following that issue, nor could find nothing on the discord server.

Can someone help me?

`

plusmoreira, 2020-06-13:
OK, now I've got the point.

Uninstalled python 3.8.3 and installed python 3.7.7 and got it working fine :)

Vik1922, 2022-09-19:
hi guys, need help with this error please. thanks in advance! 
using emotiv insight- consumer 
windows 10pro- 
intel i3 64bit
python- 3.7.7 32bit
**********
error:
C:\cykit\Py3>c:\Py372\python.exe .\CyKIT.py 127.0.0.1 5555 4
Error in file: .\CyKIT.py >>>
CyKITv2.Main() : NameError: Exception in line: 950, message: name 'eegDLL' is not defined
Traceback (most recent call last):
  File ".\CyKIT.py", line 250, in <module>
    main(1)
  File ".\CyKIT.py", line 152, in main
    headset = eeg.EEG(MODEL, cy_IO, parameters)
  File "C:\cykit\Py3\eeg.py", line 776, in __init__
    self.myKey = self.Setup(model, config)
  File "C:\cykit\Py3\eeg.py", line 950, in Setup
    eegDLL.get_bluetooth_id.restype     =  c_wchar_p  # Result   type: C-style character pointer.
NameError: name 'eegDLL' is not defined

 ) WARNING_) CyKIT2.main E1: name 'eegDLL' is not defined
Traceback (most recent call last):
  File ".\CyKIT.py", line 250, in <module>
    main(1)
  File ".\CyKIT.py", line 152, in main
    headset = eeg.EEG(MODEL, cy_IO, parameters)
  File "C:\cykit\Py3\eeg.py", line 776, in __init__
    self.myKey = self.Setup(model, config)
  File "C:\cykit\Py3\eeg.py", line 950, in Setup
    eegDLL.get_bluetooth_id.restype     =  c_wchar_p  # Result   type: C-style character pointer.
NameError: name 'eegDLL' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ".\CyKIT.py", line 265, in <module>
    mirror("Error # " + str(list(OSError)))
TypeError: 'type' object is not iterable

warrenarea, 2022-09-19:
okay, i think i fixed this.
you can re-download the project and try again.
sorry for the inconvenience.

Vik1922, 2022-09-20:
> okay, i think i fixed this. you can re-download the project and try again. sorry for the inconvenience.
no issue!
just tried the browser option -worked perfect! next will try with openvibe and update!  
much appreciated!

Vik1922, 2022-09-20:
> okay, i think i fixed this. you can re-download the project and try again. sorry for the inconvenience.

openvibe works too! 
but getting -device drift too high- any suggestions? 
also emotiv -insight has only 5 channels -can i change the number of channels to 5 and rename them?

thanks

[image]

Vik1922, 2022-09-24:
hi all, need guidance from anyone who has successfully processed the raw EEG Cykit output file.
would like to know the ideal tool (openvibe or matlab or anyother-pl do mention) to process the raw EEG Cykit output file "EEG_Recording.csv" file!
i tried with openvibe -basic scenario: csv reader->signal display & matrix display =tried both comma/semicolon separators - no success! 
if anyone knows any particular scenario wherein i can get the required info kindly share. thanks in advance!!