CymatiCorp/CyKit #21: [WARNING] Something bad happened in the loop callback, stopping the acquisition

mspr666, 2020-07-06:
hello, I am a new student studyding EEG, I install cykit and opvibe, but I can't acquire the datas from emotiv Epoc+, which gives warrinng " Something bad happened in the loop callback, stopping the acquisition ",could you give me some suggestions?

warrenarea, 2020-07-06:
can you please provide screenshots or more information concerning the errors.

as well as what command you used to run CyKIT.

mspr666, 2020-07-07:
ok,I will provide it to you immediately，thank you very much. I just now about to give you in github,there are four photos in the following

mspr666, 2020-07-07:
could you see these pictures that I gives above

mspr666, 2020-07-07:
C:\CyKit-master\Py3>c:\Python372\Python.exe .\CyKIT.py 127.0.0.1 5151 6 openvibe+generic+nocounter+noheader+nobattery+ovdelay:100+float+ovsamples:004
> USB Device (No Additional Information)
> USB Device (No Additional Information)

> Found EEG Device [EEG Signals]

> USB Device (No Additional Information)
══════════════════════════════════════════════════

> Listening on 127.0.0.1 : 5151
> Trying Key Model #: 6
(-) Connecting . . .
(+) Connected.

 ░░░ Device Interference or Turned Off ░░░

*** Reseting . . .
> USB Device (No Additional Information)
> USB Device (No Additional Information)

> Found EEG Device [EEG Signals]

> USB Device (No Additional Information)
══════════════════════════════════════════════════

> Listening on 127.0.0.1 : 5151
> Trying Key Model #: 6
(-) Connecting . . .

mspr666, 2020-07-07:
[  INF  ] Created kernel loader, trying to load kernel module
[  INF  ] Kernel module loaded, trying to get kernel descriptor
[  INF  ] Got kernel descriptor, trying to create kernel
[  INF  ] Adding kernel configuration file [C:/Program Files/openvibe-2.2.0-64bit//share/openvibe/kernel/openvibe.conf]
[  INF  ] Logging to 'C:/Users/Panrui/AppData/Roaming/openvibe-2.2.0/log/openvibe-acquisition-server.log'
[  INF  ] Adding plugins from [C:/Program Files/openvibe-2.2.0-64bit//bin/openvibe-plugins-*-stream-codecs*.dll]
[  INF  ] Added 20 plugin object descriptor(s) from [C:/Program Files/openvibe-2.2.0-64bit//bin/openvibe-plugins-sdk-stream-codecs.dll]
[  INF  ] Loading plugin: ExternalStimulations (deprecated)
[  INF  ] Loading plugin: TCP Tagging
[  INF  ] Loading plugin: LSL Output
[  INF  ] Loading plugin: Fiddler
[  INF  ] Connecting to device [Generic Raw Telnet Reader]...
[  INF  ] Connection succeeded !
[WARNING] Something bad happened in the loop callback, stopping the acquisition
[WARNING] Something bad happened in the loop callback, stopping the acquisition
[WARNING] Something bad happened in the loop callback, stopping the acquisition

warrenarea, 2020-07-07:
try modifying the preferences to the port number 5151

mspr666, 2020-07-07:
there is nothing, but I really don't know the reason why I can't gain the same result with you

warrenarea, 2020-07-07:
the parameters you used to run look correct.   i see you were getting device interference, 
you might have to try to charge the headset. 

I'd first verify you are getting the desired data being output:

c:\python27\python.exe .\CyKIT.py 127.0.0.1 54123 6 outputdata 

and verify the first group of numbers are in sequence 0 - 127

if it's not, you may have the wrong headset model #, and have to try 7 instead.

also...  i couldn't see any pictures.  i would need to see what you entered for the
acquisition server, to verify if that part is correct.

mspr666, 2020-07-07:
ok，thank you very much，I will try again tomorrow. thanks ,thanks,thanks------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;CymatiCorp&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年7月7日(星期二) 晚上11:13
收件人:&nbsp;&quot;CymatiCorp/CyKit&quot;<CyKit@noreply.github.com&gt;;
抄送:&nbsp;&quot;mspr666&quot;<2426001123@qq.com&gt;;&quot;Author&quot;<author@noreply.github.com&gt;;
主题:&nbsp;Re: [CymatiCorp/CyKit] [WARNING] Something bad happened in the loop callback, stopping the acquisition (#21)