CymatiCorp/CyKit #26: About Epoc X

batulaiko, 2020-10-20:
How Can I connect Epoc X via Cykit? What is the Key Number of it?

Have a nice day.

yxbu, 2020-10-24:
i have the same problem do you find any solution???

batulaiko, 2020-10-24:
> i have the same problem do you find any solution???
I asked EMOTIV itself. YOu choose 16-bit mode in EMOTIV APP. Then ,in CyKit, KEy is 6 I guess. I get data from it but the data is not correct in my opinion.

warrenarea, 2020-10-25:
Check with me on discord.  Discord link on this repository.

closed 2021-11-10 by warrenarea

dongchaoqun95, 2023-07-21:
May I ask if Epoc X can be supported now? I have encountered the same problem, the process can run smoothly, but the waveform is not very correct

dongchaoqun95, 2023-07-21:
@batulaiko @yxbu @warrenarea May I ask if Epoc X can be supported now? I have encountered the same problem, the process can run smoothly, but the waveform is not very correct