CymatiCorp/CyKit #27: why i can't see the signals in the html?

Minimize98, 2020-11-05:
I tried to connect epoc flex, but failed. 
Then I tried epoc insight, it shows that the status is connected, the controller id insight, but i can't see the signals.
I entered the following in CMD：
CyKIT.py 127.0.0.1 12991 4 info+confirm

Could you help me find out what's the problem? Thank you very much!!!

Solannae, 2020-11-05:
I am facing the same issue. I remember it working properly 3 weeks ago, when I first used cykit.
Data is properly extracted when using noweb, however when connecting the web interface to cykit, every sensor turns to green but no data is displayed at all.

Running with the following options:
CyKit.py 127.0.0.1 54123 7

Minimize98, 2020-11-05:
> I am facing the same issue. I remember it working properly 3 weeks ago, when I first used cykit.
> Data is properly extracted when using noweb, however when connecting the web interface to cykit, [...]

Did you connect to openvibe 3 weeks ago? I can't even find a EPOC driver in openvibe acquisition server...How should I do to install it?

warrenarea, 2020-11-05:
its a signal averaging gltich. try alternating the Format values on the left side of the webpage.

Flex is not yet supported, but work has been done with it. may check out our discord. (link on cykit page)

JolyneOY, 2020-11-08:
> its a signal averaging gltich. try alternating the Format values on the left side of the webpage.
> 
> Flex is not yet supported, but work has been done with it. may check out our discord. (link on cykit page)

THANK YOU, warren!!! I made it!!!
I still couldn't see signals in the web when I try the +openvibe command in the cmd, but when I open the openvibe designer and used the siganl display, I could see the 4 channels' signals!!!
Looking forward to your update on Flex!!!!
Have a nice day  ; )

closed 2021-11-10 by warrenarea