CymatiCorp/CyKit #31: Cykit doesn't show me the waves

CarlosA-Mtz, 2021-11-12:
I recently updated the headband but Cykit no longer showed me the waves, what can I do to fix it?

warrenarea, 2021-11-19:
Can you explain a little further?

What type of headset?
How did you update it?  
Has the headset been charged recently?

CarlosA-Mtz, 2021-11-26:
Hi, my headset is an Emotiv epoc+, and to use the emotiv BCI the emotiv software required an update as well as the headset, since then I got issues with cykit. Regarding  to the battery the headset was fully charged.

warrenarea, 2021-12-01:
yes, this may be an issue.

The EPOC+ firmware has been updated from 0x625 to 0x626
I have not yet tested out this firmware update to see what was changed.

Have you tried the 2 different keys for 128hz and 256hz?  
I think its keymodel 4, for 128hz
and keymodel 6 for 256hz

python.exe .\cykit.py 127.0.0.1 54123 4 outputdata+noweb+pywinusb
python.exe .\cykit.py 127.0.0.1 54123 6 outputdata+noweb+pywinusb

and if you could give a screenshot of your results, to see what we're working with?

warrenarea, 2021-12-09:
so i had some time to look over the code,
and i think there were some changes to the format, 
but i think they should be easy to fix.

also, to change the "device mode" (ie. 128hz to 256hz),  they made some
changes, so i'll have to create code to adjust for that.

closed 2022-03-10 by CarlosA-Mtz