CymatiCorp/CyKit #32: About developing the adapted version of EPOC X

Himhaoer, 2021-11-22:
Hello, I really admire the work you have done, and I‘m about to use the project you have developed.
I found that the device I bought was EPOC X, which is the unsupported version you marked in the documentation. However, I can still read the data, but the data seems to have some problems.
I wonder if you have any plans to continue updating the EPOC X version?
Thanks in advance!

warrenarea, 2021-12-01:
Not having an Epoc-X to test with, has stalled my efforts with it. 
Otherwise i'd like to continue to support it.

dongchaoqun95, 2023-07-21:
@warrenarea @Himhaoer May I ask if Epoc X can be supported now? I have encountered the same problem, the process can run smoothly, but the waveform is not very correct

Himhaoer, 2023-07-21:
您好：              来信收悉，不日拜读。              顺祝：工作商祺。