CymatiCorp/CyKit #36: Possibility of development in pure bluetooth

aliassar, 2022-09-11:
Hi,

I was wondering if it's possible to access raw data through bluetooth. That way we can also connect to headset with smartphones. Otherwise can you recommend another way to connect with smartphone. 
If there's possibility I'm willing to put time on it. Just let me know the steps. 

Thanks,
Ali Assar

warrenarea, 2022-09-13:
Eventually i'd really like to have infrastructure for smartphones.

Currently though, i'm stretched a little thin with trying to also get it 
working with Linux.   I rarely use a smartphone, so i'm afraid i can't 
offer to guide you.    a good  first step would b e to create an 
environment to test programs out on your phone.

aliassar, 2022-09-13:
Well, I read your specification in the Discord channel partially. Maybe that can help l. All I need is the blutooth decryption mothod since it's little bit hard to understand objective C.
That was kind of you to spent time on this.
Thanks