CymatiCorp/CyKit #40: Error when trying to connect EMOTIV Insight via Bluetooth

penco193, 2023-10-11:
Hi, 
I have an EMOTIV Insight headset (first model in white colour) and i would like to access the data by using CyKIT and connecting via BT (since my device doesn't include any USB dongle). However, after installing CyKIT for Windows and executing the command `python cykit.py 127.0.0.1 54123 4 info+verbose+bluetooth=XXXXXXXX` (with `XXXXXXXX` the bluetooth id for the headset), the following error was reported:

[image]

After looking at the code for file eeg.py, i found that the error was at line 150, so i modify the code from:

[image]

to:

[image]

So, after that modification, the command ran with any other error:

[image]

However, after opening file Web\CyKIT.html from Firefox, and connecting to the headset, none the EEG nor Gyro displayed data make sense.

[image]

[image]

I would appreciate any help in order to try solving this issue. Thanks.

warrenarea, 2023-10-11:
try changing the 'EEG resolution Slider'  to the left.

also try changing the "Convert Format" in the browser, on the left side. 

if those 2 don't work,  try adding the config flag:  +format-3

penco193, 2023-10-11:
Hi @warrenarea ,
thanks a lot for your support. I've tried both the convert format in the browser and the +format-3 flag, but at first it seems that none of those solves the problem, since, i.e. regarding the gyro data, i still received the displayed data you can see next.

[image]

I hope you can help me. Thanks in advance.

penco193, 2023-10-11:
Hi again @warrenarea ,
looking at the code in file `eeg.py`, (lines 1515 to 1524) i observed that, for the EMOTIV Insight headset connected via bluetooth with no USB dongle, the decription process is performed on `task[0:16]`, but from the 2 x 20bytes data packets received from the headset for each sample via bluetooth, the bytes `task[0]` and `task[1]` correspond to the counter/battery and sequence (1 or 2) values, respectively, whilebytes ` task[18:19]` are always `0x0000` ... so is this code correct for decripting the received data packets and creating the decripted data for this particular case? 
```
                    if eeg_driver == "bluetooth":
                        task = tasks.get()
                        decrypted = cipher.decrypt(task[0:16])
                        sel_decrypt = decrypted[0:1] + decrypted[1:16]
                        if BTLE_device_name == "Insight":
                            data = task[19:20]  \
                            + sel_decrypt \
                            + task[16:17] \
                            + task[17:18] \
                            + task[18:19]
```
Thanks for your support.

penco193, 2023-10-17:
Hi @warrenarea , is the referred code correct "as is"? In addition, just in case it can be useful, sas you can see on the attached images, the battery value shown after connecting to the headset is N/A,  ... any help will be appreciated. Thanks.